TheJoshW/klipper #29: Make DHT22 an MCU Sensor

TheJoshW, 2021-06-06:
Per comment in PR #4320:

_Thanks. However, the convention is to read sensor input from the "mcu" code and not directly from the host hardware. There was a similar discussion on this at PR #3462 . (And it may be possible to extend te ds18b20 code for the dh22 functionality.)

This distinction is also important because there are reports that the dht22 kernel code blocks the calling process during reads. The host code can't tolerate blocking reads - it could cause subtle failures.

-Kevin_

verm, 2022-12-30:
Hello, did you get anywhere on making this an MCU sensor?  Thanks.

TheJoshW, 2022-12-30:
Hi,No I haven’t had time sorry. Feel free to give it a try.

Adib-C, 2024-09-10:
Hello, any update on this?

TheJoshW, 2024-09-10:
No updates yet. If anyone has done this already please make the change. Otherwise I will get to it when I get time.

Adib-C, 2024-10-16:
Would be great if you do, Thanks.